dubstar-04/FeedsAndSpeeds #6: Blank FreeCAD Addon menu...occasionaly

spanner888, 2021-02-15:
While testing FC pull request https://github.com/FreeCAD/FreeCAD/pull/4387, found the Addon menu text sometimes goes blank. mlampert has suggested this might be an issue with this addon. I was thinking it might have been an issue with the other features being en/disable through that pull request.

My original issue notes:
    Path - Path Addons menu text is sometimes blank ...ie there is an empty/blank as frist item below menu seperator. Note in this FC config, I have dubstar-04 Feeds and Speeds addon installed https://github.com/dubstar-04/FeedsAndSpeeds. Hovering over the blank menu position, does cause the flyout menu to appear with the Feeds and Speeds addon shown.

Used path pref, to disable experimental features, saved, restarted FC and menu line blank again, and captured image, attached.
Can mostly replicate without restarting FC, just by toggling Path pref, exp features on/off. Mostly means start FC & load APath WB. Check path menu (usually correct), toggle the pref and menu item text goes blank. Toggle exp pref again (always with an OK to save) and the menu text may or may not change... I have not go to the bottom of the behaviour. eg I have seen the menu text blank survive a FC close/repopen (ie not present both before closing and still not present on reopening/restarting FC).

[image]

dubstar-04, 2022-01-23:
Is this still happening?

dubstar-04, 2022-01-24:
This only happens for me if I change the language and use FreeCAD without restarting FreeCAD.

spanner888, 2022-01-24:
Yes still happens, but for me it is fairly rare.

I have not been able to work out what triggers it.