dubstar-04/FeedsAndSpeeds #13: incorrect displayed values for Tool Flutes & chipload

spanner888, 2021-07-14:
If in FC the ToolController Toolbit Flutes is 0, GUI opens showing value of "1". Similarly Chipload shows 0.020 when TC.Tool.Chipload = 0.

The chipload issue has been staring me in the face for long time, as there is already a print showing that.

To reproduce: Create a new doc, new part cube, cut a cylinder, new job using cut, without using template and without using any existing tools. Only using default TC/TB. examine TC & TB props in FC path and values shown in FS calc.

dubstar-04, 2022-01-24:
I don't think it makes sense to support cutting tools with 0 flutes. As you need at least 1 flute to cut?

dubstar-04, 2022-01-24:
Happy for you to change my mind on this if there is a use case.