dubstar-04/FeedsAndSpeeds #15: decimal point gets messed up by factor 10

CanisLupus-42, 2021-07-27:
when starting the Feeds and Speeds Addon, the diameter value is exactly 10 times higher, than specified in the ToolController.

[image]

OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.25306 (Git)
Build type: Release
Branch: master
Hash: 5f5b3d9b67136edda1f6738f1a83154990f7839d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

Also tested with the stable release of FreeCAD 0.19 -> same issue

CanisLupus-42, 2021-07-27:
Just found another occurrence of this problem: it seems to be related to the decimal seperator. As I have a german Windows underneath, the decimal sperator is a ',' not a '.'
When entering the stepover or stepdown with a ',' , everything works fine. When entering with a '.' as decimal the following happens:
input -> field value, after exiting the field
12 -> 12,00
12,0 -> 12,00
12.0 -> 120,0
0.12 -> 12,00
0.12 mm -> 12,00 mm
It seems, as if the '.' is simply removed and the result is displayed in the local decimal format, in this case using a ','

Probably the same for the tool diameter, as the screenshot above shows the tool diameter as '6.0 mm' in the "Report view". Intrestingly the chipload is not affected by this phenomenon.

dubstar-04, 2022-01-24:
Does this still happen if you change the language settings in FreeCAD? 

[image]

CanisLupus-42, 2022-01-25:
It was already set to english, so the problem persists.
[image]

dubstar-04, 2022-01-25:
> It was already set to english, so the problem persists.
> If I can provide any other helpful information, I would be happy to do so.

Thanks for the quick reply. 
I think the Qt framework reads the locale info from the operating system. What language is your OS using?

CanisLupus-42, 2022-01-25:
Changing the FreeCAD Language to German does not help, even with an additional restart of FreeCAD.
But your hint with the locale information from the operating system did the trick. At least somehow. When I change my system-culture (PowerShell: Set-Culture en-EN) it works. But then I get problems with other software (e.g. MS Excel), therefore it is no real solution, sadly.

dubstar-04, 2022-02-01:
Future ref: 
https://forum.freecadweb.org/viewtopic.php?f=10&t=18833&start=10